ob-f/OpenBot #248: [help wanted] Where can I get the pre-trained models (or detailed instructions)?

omer-re, 2022-01-07:
Hi,
I am building my OpenBot, 
I was able to get to the stage where it drives using the remote control app and I am now trying to get it to the next step of following objects and autonomous driving.

I couldn't fully understand how to do what described on the demo video on the object detection and following [as shown on the demo video  #](https://i.imgur.com/NE6WDXD.jpeg).
**Is there any place I can find instructions for it? link, video, tutorial etc,.**

I have also tried to understand if there are existing pre-trained policy files that I can use 
[(again, as shown on the demo video).](https://i.imgur.com/IIoa2E7.png)

I know it can be fun to train it on my own, but I would first like to have some baseline, and I couldn't find some nice guide or tutorial for it, not the files for it.

I would really appreciate some directions 😁

thias15, 2022-01-07:
For autonomous driving you need to collect data and train your own policy since its specific to your environment. You can try the one provided in the Autopilot fragment but it probably won't work unless your corridors looks very similar.

thias15, 2022-01-07:
For object tracking just open this fragment and activate the auto mode.
[image]

thias15, 2022-03-08:
Please check out these excellent videos:
Person Following: https://youtube.com/playlist?list=PLNKFHX5MRn501oWvPbKzP1zkcqhLU5TOh
Policy Learning: https://youtube.com/playlist?list=PLNKFHX5MRn5233AyCWhcn71JdB9qIEa-E

closed 2022-03-08 by thias15