ob-f/OpenBot #268: controller app is crashing

robotixdevteam, 2022-03-22:
Today, controller app has crashed which has been downloaded from QR code. previously app is working we have faced bluetooth connection stability issues, is there any update for this issues???

thias15, 2022-03-22:
Can you please provide further details. @izivkov can you help @robotixdevteam to figure out a solution?

robotixdevteam, 2022-03-23:
sorry, now we have removed all the previous apps and installed new ones again now. They are installed.
But now we are facing mobiles are not getting connected to each other?

izivkov, 2022-03-23:
If you are talking about the controller phone not connecting to the robot
phone: make sure both phones are on the same LAN, not using data
connection. Make sure they are using the same router, i.e. not on a
University WiFi network, or something like that. Then try a few times,
sometimes they don't connect the first time.

If still doesn't work, please provide more information on what is going on
and your setup.

Thanks

robotixdevteam, 2022-03-23:
These are the steps we followed for connecting the mobiles.
connected the mobile to bot and open the apps in the mobile and change the control in viewer app to phone and have wait for some time. one more thing we connected the mobiles to same wifi connection. We have tried this many times still they are connected. I want to know whether the procedure for connecting is right or we are missing something to change?

robotixdevteam, 2022-03-23:
hi, another thing i would like to know is why you are using the arduino 33 iot, as the mobile has both wifi and bluetooth?

izivkov, 2022-03-23:
Ok, looks like you are doing the right steps, but also try the following:

1. Restart the Controller App on the right phone each time you are trying to connect
2. On the Main software, switch control between Gamepad and Phone a few times, The connection is getting initiated when the "Phone" control mode is selected. 
3. Also, from the main screen, selected "Free Form" icon, and then switch the "Controller" icon (bottom left button) a few times.

Hope this helps

izivkov, 2022-03-23:
https://user-images.githubusercontent.com/4115966/159732255-eadd8d57-75c6-4a7a-ba94-0a46891d1a87.mp4

robotixdevteam, 2022-03-24:
is the phones are connected using the wifi or bluetooth?

thias15, 2022-03-28:
> ng about the controller phone not connecting to the robot phone: make sure both phones are on the same LAN, not using data connection. Make sure they are using the same router, i.e. not on a [...]

This is just since the Arduino 33 IOT can be used also for other things. But it can work with Arduino Nano as well.

thias15, 2022-03-28:
There should not be a specific requirement for which phones can be used, but we did not test with the phones you mentioned above. You closed the issue, so I just wanted to check if the issue is resolved now from your end?

robotixdevteam, 2022-03-28:
No, the issue was not solved we are trying to understand where exactly is the problem if you have any ideas on how to solve it share with us?

robotixdevteam, 2022-03-28:
we wanted to open a new issue regarding it so we closed it

izivkov, 2022-03-28:
As I discussed with @robotixdevteam team last week, the problem seems to occur in only one of the phones: Redme. Since we cannot test all phones, we need to work together with the team on this issue. Maybe the team can provide some logs from this phone, to see if there are any crashes or what the problem is.

robotixdevteam, 2022-03-30:
we tested the node-js .while testing node-js, we came across certain issue of connectivity with some windows laptops. It is working good with raspberry pi and ubuntu os. currently, Have you tested the node-js with windows laptops. Right now, we are testing the python controller. Oh yes, i forgot to mention the main issue connectivity among two mobile phones that issue has not been solved from our side. Is there any updates regarding it.

robotixdevteam, 2022-03-30:
we tested the python code. some windows laptops there is connectivity issue and rest even though it is connected and popup-dialogue can  be seen but the video is not streaming through both RTSP and WEBRTC. We have not tested the python code as well as node-js in real time.

izivkov, 2022-03-30:
> we tested the python code. some windows laptops there is connectivity issue and rest even though it is connected and popup-dialogue can be seen but the video is not streaming through both RTSP and WEBRTC. We have not tested the python code as well as node-js in real time.

The Python controller works with RTSP video only.
Check and disable any firewall and virus protection on your Windows machines, maybe this will help.
The NodeJS and Python controllers have been developed on and for Linux-based OS's so far. It would be great if we can get them working on Windows, though.

robotixdevteam, 2022-03-31:
Prevously it connected to windows without video streaming but after disabling the protection and firewall it is not even getting connected.

thias15, 2022-04-28:
Any update on this?

izivkov, 2022-04-29:
This looks more like a configuration or a firewall issue. I think the same software works on some systems, but not on others. The Python controller was tested on Linux, and I left is as a "TO DO" to test on Windows, and recommend configuration. As of now, we cannot say that the Python controller is tested or supported on Windows. Since I am not running Windows on any of my machines, somebody can test it and let us know.

closed 2022-04-29 by thias15