ob-f/OpenBot #275: why openbot app is not supported all other android versions?

Srini4063, 2022-04-06:
**Describe the problem**
A clear and concise description of what the problem is or what information you are looking for.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows, Linux, iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. Samsung S20]
 - OS: [e.g. Android 11]
 - Browser [e.g. stock browser, chrome]
 - Version [e.g. 0.1.1, master]

**Additional context**
Add any other context about the problem here.

Srini4063, 2022-04-06:
why  openbot was not supported all other android version ?

thias15, 2022-04-06:
What do you mean?

Srini4063, 2022-04-06:
Other android phones are not communicated well what can I do?

quentin-leboutet, 2022-04-07:
Hi [Srini4063](https://github.com/Srini4063), could you give us more details about the problem you are currently experiencing? To be able to support you, we need to understand the context in which you encountered this issue, so that we can reproduce it. For instance, which version of Android are you using? What is your operating system? What is your version of Android Studio? What is the observed behavior? Which protocol did you follow to get this behavior? Which features seem to be affected? Do you have any error messages? This is really important.

thias15, 2022-04-22:
Closing due to inactivity.

closed 2022-04-22 by thias15