ob-f/OpenBot #284: video stream error

alstjr510, 2022-05-05:
**Describe the problem**
A clear and concise description of what the problem is or what information you are looking for.
It connects when I run (python keyboard-pygame.py --video), but the video doesn't come out of my laptop!

Can't read stream is an error message when running ( python keyboard-pygame.py --video)

I want to know the camera address information of the cell phone connected to the open bot.(On Openbot Application)

**To Reproduce**
Steps to reproduce the behavior:
1.conda activate [your env]
2.cd Openbot
3.cd controller
4.cd python
5. python keyboard-pygame.py --video

**Expected behavior**
A clear and concise description of what you expected to happen:
I can see the video of the cell phone camera connected to the open-bot app on my laptop.

**Screenshots**
If applicable, add screenshots to help explain your [problem.]

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser :  chrome
 - Version 18.04

**Smartphone (please complete the following information):**
 - Device: Samsung S10
 - OS: Android 12
 - Browser :chrome
 - Version : I didn't understand what version you were talking about.

**Additional context**
Add any other context about the problem here.

thias15, 2022-05-05:
Did you choose RTSP in the settings of the OpenBot app?

alstjr510, 2022-05-05:
> Did you choose RTSP in the settings of the OpenBot app?

Yes, I know that there are WebRTC and RTSP in Stream mode. And I set it to RTSP.

thias15, 2022-05-11:
@izivkov any ideas?

alstjr510, 2022-05-23:
Hello! I'm working on your open-bot project. There is one question.
Even if I update the app, Point Goal Navigation is not on the interface, but hasn't the function been added yet?

closed 2022-06-02 by thias15

alstjr510, 2022-10-11:
Yes, I know that there are WebRTC and RTSP in Stream mode. And I set it to RTSP.