ob-f/OpenBot #286: Fast tracking a single person.

emo8899, 2022-06-28:
**Is your feature request related to a problem? Please describe.**
When I choose to follow people, if there is another person along the way, the car will randomly choose to follow one of them.

**Describe the solution you'd like**
Fast tracking a single person.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
https://github.com/foolwood/SiamMask

thias15, 2022-07-19:
Yes, this is expected since it is currently using an off-the-shelf object detector. Would you like to contribute to the integration of an object tracker?

emo8899, 2022-07-19:
> Yes, this is expected since it is currently using an off-the-shelf object detector. Would you like to contribute to the integration of an object tracker?

I'm really sorry. I really want to help you finish this work, but my professional ability is limited.

thias15, 2023-02-14:
On hold for now.

closed 2023-02-14 by thias15

yudelex, 2024-02-23:
Is object tracking implemented in "Detector.java" file?