ob-f/OpenBot #290: Do you plan to support multicopters?

atolkachiov, 2022-07-25:
As far as I understand, there are plenty of pros to use smartphones as onboard computer for multicopters, for starters at least. 
* sensors to maintain stable flight 
* independent power source 
* 4g connectivity 
* ML-grade performance 
* great price 
* availability (which is often a blocker if you are going to make a build using some SBCs) 

Do you have it on your roadmap? Looking for contributors?

thias15, 2022-07-26:
Hi @atolkachiov. We have considered it before, but there were a few concerns.
(1) People may be more hesitant to put their phone on a flying platform
(2) On quadrotors weight is a major concern and phones have quite a bit of extra weight due to battery, screen and chassis
(3) Integration of a full flight control stack in Android requires a lot of work

As a result the first priority has been ground vehicles. That being said, contributions would be most welcome.

atolkachiov, 2022-07-27:
Probably it would be easier to rely on Ardupilot compatible SoC as a flight controller and have smartphone as an auxiliary [companion computer](https://ardupilot.org/dev/docs/companion-computers.html) module focused on ML tasks and autonomous flight. As Ardupilot supports Odroid boards which run Android, adding multirotors supports may essentially be introducing MAVLink protocol bridge API available for the user app. 

I understand all the concerns, but professional turnkey solutions cost a fortune and even then often they are not available because of chip shortage. Having some extra weight onboard sounds like an acceptable tradeoff If you can get powerful GPU + quality cameras + lidar + cellular connectivity for a few hundreds dollars. Good for learning and prototyping.

thias15, 2022-10-13:
Hi @atolkachiov. I'm closing this for now since this is not a high priority on our roadmap and we have very limited bandwidth. However, if you would be interested to work on this we would be happy to support as much as we can!

closed 2022-10-13 by thias15