ob-f/OpenBot #293: firmware

mjanddy, 2022-08-29:
when I upload the firmware,it is error which that is some fuction is declared after. How should I fix it?
[image]

quentin-leboutet, 2022-08-29:
Hello mjanddy, 

thank you for using OpenBot !
Could you give us more information about the context in which you encounter this error so that we can reproduce it ?
For instance, on which board are you trying to flash this code ? 
Did you modify the firmware ? In which way ? 

Thank you 

Quentin

quentin-leboutet, 2022-08-30:
Strange... I cannot reproduce your bug. Are you using the last IDE version ? What are your exact settings ?

mjanddy, 2022-08-30:
> Strange... I cannot reproduce your bug. Are you using the last IDE version ? What are your exact settings ?

I use arduion ide 1.8.19; just as  github setting

mjanddy, 2022-08-30:
> > Strange... I cannot reproduce your bug. Are you using the last IDE version ? What are your exact settings ?
> 
> I use arduion ide 1.8.19; just as github setting

> Strange... I cannot reproduce your bug. Are you using the last IDE version ? What are your exact settings ?

 and now  I moved the position of the function, but I'm not sure if it's correc

quentin-leboutet, 2022-08-30:
Did you try compiling the firmware using the Arduino [cloud IDE](https://create.arduino.cc) ?

mjanddy, 2022-08-30:
> the Arduino [cloud IDE](https://create.arduino.cc/editor) ?

ok,I will try it ,thanks!

mjanddy, 2022-08-30:
> Did you try compiling the firmware using the Arduino [cloud IDE](https://create.arduino.cc/editor) ?

It is ok,thank you!

closed 2022-08-30 by mjanddy