ob-f/OpenBot #295: Where precisely is ESP32 being used in the DIY bot ?

mugdhapendse, 2022-10-13:
I am unable to see ESP32 in the wiring diagram of DIY bot. 
So, when should the steps mentioned in the ESP32 be implemented ? Is it an alternative to arduino nano ?

mugdhapendse, 2022-10-14:
Alright, understood. A couple more questions though : 

1] For the current DIY version, are we not enabling bluetooth and wifi feature ? What is the exact purpose of these features ?

2] In the overall project,  we need only 1 smartphone to be mounted on the bot, right ?

mugdhapendse, 2022-10-14:
@thias15 Could you assign this to me ?

I would firstly like to contribute to the the documentation of "Arduino Firmware" that the ESP32 kit is only for the RTR version to avoid any further confusion of other users.

I also plan on using only ESP32 in the DIY version and sharing the project here eventually.

Do let me know if this is possible.

thias15, 2022-11-29:
@mugdhapendse your contributions will be most welcome. The BT on the ESP will allow wireless connection between microcontroller and Smartphone App (WIP). Having WiFi on the ESP32 is a bonus that can enable further applications in the future, e.g. send commands over network without a phone.

marshalab, 2024-11-22:
[openbot_sch.xlsx](https://github.com/user-attachments/files/17873453/openbot_sch.xlsx)
Hello/ I am collect some data from source in the table. This is may help you.

In my research version, I want to use the M5stask Lego modules that have been lying in a drawer for a long time. This should make a great build for experimentation. Already found two NXT motors with built-in encoders, smartphone. Now it is necessary to establish order and find the module itself esp32 m5stask.