ob-f/OpenBot #297: Shared baseline dataset to play with Driving Policy

mnrozhkov, 2022-10-19:
**Is your feature request related to a problem? Please describe.**
New users may be interested in playing around with data and train models for Driving Policy before they built own OpenBot and setup data collection.  It would be cool to have an open dataset for this. 

**Describe the solution you'd like**
- Prepare a baseline dataset and upload to cloud storage, with an public read-only access. 
- Add a link and instructions to  Driving Policy readme

**Describe alternatives you've considered**
- upload this data to Kaggle  (this may bring more DS users into the project) 

**Additional context**
This is my personal pain at this moment. I don't have my OpenBot assembled, and it may take few weeks before I could get one. 
If someone has datasets that can be opened, I would happy to help preparing shareable link and updates in README.

thias15, 2022-11-29:
We are working on a simulator that can be used for data collection and to try out the OpenBot training pipeline without a physical robot. Stay tuned!

thias15, 2023-03-23:
Sorry for the late response. You can check out the simulator [here](https://github.com/isl-org/spear/tree/main/examples/imitation_learning_openbot).

closed 2023-03-23 by thias15