ob-f/OpenBot #335: Is there already an existing .java file for using YOLOv7 tiny in OpenBot?

RoBoFAU, 2023-02-13:
Do you have any idea which parameter/function I have to change to deploy custom YOLOv7 tiny with the OpenBot?

Any idea is helpfull! :-)

quentin-leboutet, 2023-02-13:
Hello RoBoFAU, 

Seems like a nice project! I'm not so familiar with YoloV7 but I think you may need to implement an extension of the [detector](https://github.com/isl-org/OpenBot/blob/master/android/app/src/main/java/org/openbot/tflite/Detector.java) class, for instance in a dedicated **DetectorYoloV7.java** file. You could use [DetectorYoloV4.java](https://github.com/isl-org/OpenBot/blob/master/android/app/src/main/java/org/openbot/tflite/DetectorYoloV4.java) or [DetectorYoloV5.java](https://github.com/isl-org/OpenBot/blob/master/android/app/src/main/java/org/openbot/tflite/DetectorYoloV5.java) as templates to be fine-tuned to the input/output shape of your data. You should also let the system be aware about your detector's .tflite file (I here assume you are using tensorflow lite) by adding the suitable fields in your [config.json](https://github.com/isl-org/OpenBot/blob/master/android/app/src/main/assets/config.json) file (you can use the previous entries in this file as a template). Also don't forget to add a "YOLOV7" field in the [model enums](https://github.com/isl-org/OpenBot/blob/135c16484e47909e9095b8dcde98955b732f82cf/android/app/src/main/java/org/openbot/tflite/Model.java#L31) as well as in the [Detector switch](https://github.com/isl-org/OpenBot/blob/135c16484e47909e9095b8dcde98955b732f82cf/android/app/src/main/java/org/openbot/tflite/Detector.java#L69).

Doing so should bring you pretty close to a working solution. Let us know if you face any issue.

RoBoFAU, 2023-02-13:
Hello!

Thanks for your answer! I already did exactly what you wrote but it is not working.
I copied the DetectorYolov5 file and tried to apply it to Yolov7 tiny. Also added the tflite file to assets/networks, config.json and the model enums and Detector switch.
I am really new to object detection/yolo topic, that's why it is hard for me to fine-tune input/output parameters that are required from the OpenBot App.

RoBoFAU, 2023-02-13:
Could you maybe help me by telling which parameters are typically the ones to change/adjust?

thias15, 2023-02-14:
Hi @RoBoFAU. One important factor is the input and output dimension. You can load your tflite file using [netron](https://netron.app/) to find out.

RoBoFAU, 2023-02-14:
> Hi @RoBoFAU. One important factor is the input and output dimension. You can load your tflite file using [netron](https://netron.app/) to find out.

yolov7tiny.tflite:
[image]

in comparison to yolov5s.tflite:
[image]

thias15, 2023-03-03:
Ok, so the input name has to be changed and also the image tensor is in a different format. You would also need to figure out how to parse the output since it's a different format.

closed 2023-05-04 by thias15