ob-f/OpenBot #336: [DIY] Nano board freeze and stop accepting commands after starting the motor with c command.

shashank734, 2023-02-14:
Hey Folks, Building DIY and Connected the Arduino nano with the studio, Uploaded the firmware it's connected and showed the data on OLED and serial monitor both. when I send the i0,1, i1,0 ... commands, it's working fine.

But when I send the c command motor starts, but OLED turns off, and it stops sending the data on the serial monitor, it stops accepting the commands need to unplug the USB and insert it again to stop the motor. Turn on/off the power switch without removing the USB. still, motor starts and not accepting commands to make it work need to remove the USB. I think the board freezes after sending motor commands. Tried with Not connected to the phone yet.

The same is happening with the phone App; after starting the motor, no other commands working need to reconnect.

PS: I observe with the phone it was accepting the instructions when the motor was at a slow speed once the motor is at high-speed everything gets freeze.

Using this board https://robu.in/product/arduino-nano-board-r3-ch340-chip-with-usb-cable-solderedarduino-nano-r3-usb-cable/

Thanks for great project :)

thias15, 2023-02-14:
Hi @shashank734. It sounds like the electric noise of the motors might be causing interference. Can you try to shield the USB cable and keep it as far as possible from the motor wires? Another reason could be a faulty board (but then it's weird why it works for low motor speeds), especially the cheap Arduino clones are not always reliable.

shashank734, 2023-02-16:
Hi @thias15, Thanks for the response will try this on the weekend and update here. Actually, it's only sometimes working at a slow speed; it worked only a few times for a few seconds than stuck.

shashank734, 2023-02-19:
Hey tried to put Arduino at distance from the motor, and then it works but after some time again freezes. Maybe an issue with Arduino but don't know why it's working sometimes by putting it at a distance.

thias15, 2023-03-03:
Hard to debug remotely. Would recommend trying a new/different Arduino board.