ob-f/OpenBot #354: Unpowered arduino nano

rfedersp, 2023-04-26:
Hi, I'm a total beginner in electronics and I'm trying to build the DIY version of OpenBot (which is a great project!). The problem is that I can't power the Arduino Nano with the voltage divider bridge... Is my assembly wrong?

[image]

thias15, 2023-04-26:
Hard to tell from the image but looks roughly ok. Can you measure the voltage with a multimeter? Note that if you built the voltage divider as per instructions and connected it to the corresponding pins the purpose is not to power the Arduino but to measure the battery voltage. The Arduino is powered via the USB connection to the phone.

rfedersp, 2023-04-27:
I measure a voltage of 0.6 but it must be normal if it is not intended to power the arduino.  
On the other hand, when I connect my phone to the arduino, no led lights up, is that normal?

rfedersp, 2023-04-27:
The problem was with my OTG cable... Everything works now! 
Thank you for your help :)

closed 2023-04-27 by rfedersp