ob-f/OpenBot #357: NodeMCU ESP32s

lechfak, 2023-05-04:
Hello.
I want to make DİY_ESP32 with NodeMCU ESP32s.
I used L298 as motor driver.

Where should I connect the CH_PWM_L1/ L2/R1/R2 and PIN_PWM_LF1/LB1... pins? I don't understand this.
When CH_PWM_L1 remains as is, all engines are running in NO_PHONE mode only. But none of them work in normal mode.

When I do the same with CH_PWM_... and PIN_PWM_.LF/LB.., only the motors on the left respond to commands. 
//PWM properties
const int FREQ = 5000;
const int RES = 8;
const int CH_PWM_L1 =13;
const int CH_PWM_L2 = 12;
const int CH_PWM_R1 = 27;
const int CH_PWM_R2 = 33
const int PIN_PWM_LF1 = 13;
const int PIN_PWM_LF2 = 12;
const int PIN_PWM_LB1 = 13;
const int PIN_PWM_LB2 = 12;
const int PIN_PWM_RF1 = 27;
const int PIN_PWM_RF2 = 33;
const int PIN_PWM_RB1 = 27;
const int PIN_PWM_RB2 = 33;

What can I do?

[image]
[image]

thias15, 2023-05-04:
Please use the latest firmware. The `CH_PWM` can just have values 0,1,2,3. The physical GPIO pins are defined as `PIN_PWM_LF1`, `PIN_PWM_LB1`, etc. Note that LF (left front) and LB (left back) are defined as the same pin since we only have two outputs - one for the left side and the other for the right side. You should connect pins 13 and 12 to IN1 and IN2 to control the left side and pins 27 and 33 to IN3 and IN4 to control the right side.

thias15, 2023-05-04:
If it works in the No_PHONE mode, it means that the wiring is correct. It seems the issue you are facing is with the connection between robot and phone then. Did you try to send commands from the serial monitor in Arduino? Does the robot info screen work when the robot is connected?

lechfak, 2023-05-05:
Hi.
1.  yes. i try to send commands from the serial monitor in Arduino. wıth c255,255. but there is no change
2. robot info screen dont work when the robot is connected
![WhatsApp Image 2023-05-05 at 14 41 33 (1)](https://user-images.githubuserconten
[image]
[image]
[image]
t.com/24843137/236449220-bd0543d9-e887-4c8c-8ffc-2f94f8914491.jpeg)
[image]

lechfak, 2023-05-09:
Hi. 
I was able to connect to the device using bluetooth connection.
It didn't happen with the USB cable at the same settings.
The robot info screen has seen the device.
but after working for a few seconds, the device turns itself off by giving this error:

Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060130  A0      : 0x800d14e9  A1      : 0x3ffcafc0  
A2      : 0x3ffc45f8  A3      : 0x00000000  A4      : 0x0000aeb4  A5      : 0x0000da80  
A6      : 0x00000020  A7      : 0x3ffc90e0  A8      : 0x800d6caf  A9      : 0x3ffcafa0  
A10     : 0x3ffc45f8  A11     : 0x3ffba660  A12     : 0x00000005  A13     : 0x00000005  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000000a  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x00000000:0x3ffcafc0 0x400d14e6:0x3ffcafe0 0x400d1586:0x3ffcb000 0x400d860c:0x3ffcb020 0x400905de:0x3ffcb040

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

hardikgarg02, 2023-09-15:
Hey! Which Android device have you used for connecting to the robot? If connecting via USB and the serial monitor is not sending commands to the robot, try using a different device for the connection.

hemangjoshi37a, 2024-10-14:
I am using the defualt firmware file and getting this error in my esp32 devkit v1 :
```
17:29:17.656 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:29:17.656 -> configsip: 0, SPIWP:0xee
17:29:17.656 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:29:17.656 -> mode:DIO, clock div:1
17:29:17.656 -> load:0x3fff0030,len:4832
17:29:17.656 -> load:0x40078000,len:16460
17:29:17.656 -> load:0x40080400,len:4
17:29:17.656 -> load:0x40080404,len:3504
17:29:17.656 -> entry 0x400805cc
17:29:18.399 -> ets Jul 29 2019 12:21:46
```