ob-f/OpenBot #377: Unable to Connect Android App Openbot to Controller App

A-janjan, 2023-10-20:
I have encountered an issue where the Openbot app is unable to connect to the Controller app. I have installed both of them and connected them to the same network, but they aren't connected to each other. Additionally, I have changed the controller type to phone.

What should I do?

Eashika2504, 2023-10-23:
Make sure to open the controller app first and then connect to OpenBot by switching the controller mode to phone. If you're still facing issues, consider switching to a different device.

A-janjan, 2023-10-23:
It didn't work for 3 devices;  Are there any logs or things that can show me what happened backstage?

hardikgarg02, 2023-12-14:
Yes, When successful connection establishes, in robot app log appears  as  "Client connected: <IPv6 link-local address of wifi>".   And if there comes any error it shows "Got exception: <error> ".

abhijeetmhatre, 2024-01-10:
I am also facing the same issue. 

The setup is pixel6a(controller) running a WiFi hotspot. Samsung Galaxy M31(robot) and also a laptop. Tried reversing the two phones. Also tried  third phone hosting the hotspot instead of one of the  two

 I am able to control the car using the python scripts. in laptop but not with the controller app.

M-S-10, 2024-02-21:
It works fine for me! Make sure that both the phones are on the same WiFi Network.

M-S-10, 2024-03-06:
Did it work? if yes then please close this issue.

A-janjan, 2024-03-07:
Yeah, but the main solution is changing the stream mode from RTSP to WebRTC. Thanks to all.

closed 2024-03-07 by A-janjan