ob-f/OpenBot #378: Running without stopping

A-janjan, 2023-10-21:
When I go to the Robot info section and then click the forward or backward command in the send command section, I see that my motors start running without stopping. Additionally, when I click stop, nothing changes.

I use POCO X3 Pro.

hardikgarg02, 2023-11-02:
Please ensure that the app maintains a stable Bluetooth connection with the robot, without any disconnections between commands.

A-janjan, 2023-11-03:
No, I didn't use Bluetooth. I just used the main application and the following sections:

[image]

A-janjan, 2023-11-07:
I did the firmware testing with an Arduino serial monitor, and at first, I sent, for example, c100,100 and then the motors started running. but after I sent c0,0 and the motors didn't stop. I don't know why motors can't be stopped.

Eashika2504, 2023-12-15:
Ensure that your motor connections are accurate, and double-check that the specified serial pin numbers in the code align with the actual hardware configuration.

A-janjan, 2024-03-02:
ok

A-janjan, 2024-03-03:
After I sent a command via a serial monitor (e.g., c50,0), the motors moved correctly, but the serial connection disconnected, and I couldn't send a new command.

A-janjan, 2024-03-27:
The main problem arises when I enable #define HAS_OLED 1 in the DIY body firmware. Consequently, other commands fail to send correctly. However, after disabling the OLED, everything is fixed.

closed 2024-05-08 by A-janjan