ob-f/OpenBot #387: When using one phone as Wifi AP and the other connect to this AP, controller app and robot app can NOT connect

yijunwu, 2023-12-09:
**Describe the bug**
When using one phone as Wifi AP and the other connect to this AP, controller app and robot app can NOT connect

**To Reproduce**
Steps to reproduce the behavior:
1. Two phones, lets say A and B. Start Wifi Access Point on A
2. On phone B, connect to that Wifi provided by A
3. Run robot App on A, select phone as control mode; Run controller App on B
4. Robot app can not find controller app (service) hence from user point of view, controller can't connect to robot

**Expected behavior**
When using one phone as Wifi AP and the other connect to this AP, controller app and robot app can connect. This will be a convinient way to allow users to use phone control mode even in out-door scenario.

**Smartphone (please complete the following information):**
 - Device: Samsung S20FE, Samsung A40S, Samsung A6S tested
 - OS: Android 10, 11
 - Version 0.7.1, master

**Additional context**
I've worked out a fix for this issue, will create PR soon.

yijunwu, 2023-12-09:
I've create a PR (https://github.com/isl-org/OpenBot/pull/388) to fix this. Please kindly review and merge.

hardikgarg02, 2023-12-15:
Its working fine for android. Can you also check same functionality for IOS application.

yijunwu, 2023-12-17:
> Its working fine for android. Can you also check same functionality for IOS application.

I'll take a look, but I'm not familiar with iOS dev, so I'm not sure how much time it'll take.

closed 2023-12-29 by sparsh3dwe