ob-f/OpenBot #411: trained policy file

A-janjan, 2024-05-06:
Hi, how can I find the policy file (.ckpt or .xml, or .tflite file)? I need it. please upload it.

Surajitinker, 2024-05-14:
If you're looking for the policy file (.ckpt, or .xml, or .tflite), you can find it in the Model Management section. There, you'll see a collection of files, including the assets file and the necessary .tflite files. Simply navigate to the Model Management tab, and you should find what you're looking for.
And If you've trained your own model and are looking for the policy file, you'll typically navigate to the directory named "policy".

A-janjan, 2024-05-18:
Thanks. but I didn't see the location of tflite files. in the model management section, we can just see models and not their locations in phone storage.

hardikgarg02, 2024-05-20:
The tflite files are downloaded from web using this json file. You can visit path for particular model and download it from there.
[config.json](https://github.com/isl-org/OpenBot/files/15373115/config.json)

A-janjan, 2024-05-21:
Yeah, thanks. but how can I get autopilot_float.tflite ? because in config.json , it doesn't have url.

hardikgarg02, 2024-06-13:
For autopilot_float.tflite :- [autopilot](https://storage.googleapis.com/openbot_models/autopilot_float.tflite)

Its all available in https://storage.googleapis.com/openbot_models/<Model_Name>

You have to just put model name and the tflite will be download.