ob-f/OpenBot #432: Upload error : exit status 2 on chip ESP32-D0WD-V3 (revision v3.0)

hexs00si, 2024-06-17:
**Describe the problem**
Despite the successful compilation of the given sketch , along with using the right board and port with all the compatible libraries , the sketch fails to be flashed on the firmware . It generates the following error message 

```
Sketch uses 1038073 bytes (79%) of program storage space. Maximum is 1310720 bytes.
Global variables use 45708 bytes (13%) of dynamic memory, leaving 281972 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.6
Serial port /dev/cu.usbserial-210
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:e3:6d:c6:93:40
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.

A fatal error occurred: Unable to verify flash chip connection (Invalid head of packet (0xE0): Possible serial noise or corruption.).
Failed uploading: uploading error: exit status 2**
A clear and concise description of what the problem is or what information you are looking for.

```
**To Reproduce**
Steps to reproduce the behaviour:
1. Go to tools and select your respective board , for me i chose the ESP32 Dev module
2. Select the appropriate port.
3. Verify your sketch , as well as the installed libraries.
4. Click on upload.

**Expected behavior**
The sketch should be successfully flashed to the connect esp32 chip.

**Desktop (please complete the following information):**
 - OS: [macOS sonoma14.5]
 - Arduino IDE [version 2.3.2]

hexs00si, 2024-06-17:
[433](https://github.com/isl-org/OpenBot/pull/433). Please review and merge

hemangjoshi37a, 2024-10-14:
I am getting this error in the serial console while uploading the default firmware to my esp32 devkit v1 :
```

17:29:17.656 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:29:17.656 -> configsip: 0, SPIWP:0xee
17:29:17.656 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:29:17.656 -> mode:DIO, clock div:1
17:29:17.656 -> load:0x3fff0030,len:4832
17:29:17.656 -> load:0x40078000,len:16460
17:29:17.656 -> load:0x40080400,len:4
17:29:17.656 -> load:0x40080404,len:3504
17:29:17.656 -> entry 0x400805cc
17:29:18.399 -> ets Jul 29 2019 12:21:46
```

please help.