ob-f/OpenBot #447: the layer MLP has never been called and thus has no defined output

sparta-2024, 2024-08-16:
**Describe the bug**
When using tensorflow 2.16 or 2.17 to do train ,there is  an  error occurred:the layer MLP has never been called and thus has no defined output.
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows, Linux, iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. Samsung S20]
 - OS: [e.g. Android 11]
 - Browser [e.g. stock browser, chrome]
 - Version [e.g. 0.1.1, master]

**Additional context**
Add any other context about the problem here.

sparta-2024, 2024-08-16:
models.py file should be added the following code:
import os
os.environ["TF_USE_LEGACY_KERAS"] = "1"

isha382, 2024-08-21:
hi @sparta-2024, the issue is of the version of tensorflow you can try with 2.9.0 version which is preferred version and can refer following documentation [policy](https://github.com/3dwesupport/OpenBot/blob/master/policy/README.md) for more information.