ob-f/OpenBot #455: getting this error in esp32

hemangjoshi37a, 2024-10-14:
```

17:28:34.751 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:28:34.751 -> configsip: 0, SPIWP:0xee
17:28:34.751 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:28:34.784 -> mode:DIO, clock div:1
17:28:34.784 -> load:0x3fff0030,len:4832
17:28:34.784 -> load:0x40078000,len:16460
17:28:34.784 -> load:0x40080400,len:4
17:28:34.784 -> load:0x40080404,len:3504
17:28:34.784 -> entry 0x400805cc
17:28:35.526 -> ets Jul 29 2019 12:21:46
```

getting this error in esp32 by uploading the default firmware arduino file without any changes .

isha382, 2024-10-14:
Hi @hemangjoshi37a, You need to change this line  if (OPENBOT == DIY) to if(OPENBOT == DIY_ESP32) in the arduino file then you can upload.
[image]

hemangjoshi37a, 2024-10-14:
now i am getting this error :
```
/home/hemang/Documents/GitHub/OpenBot/firmware/openbot/openbot.ino: In function 'void setup()':
/home/hemang/Documents/GitHub/OpenBot/firmware/openbot/openbot.ino:906:3: error: 'ledcSetup' was not declared in this scope; did you mean 'ledc_stop'?
  906 |   ledcSetup(CH_PWM_L1, FREQ, RES);
      |   ^~~~~~~~~
      |   ledc_stop
/home/hemang/Documents/GitHub/OpenBot/firmware/openbot/openbot.ino:912:3: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
  912 |   ledcAttachPin(PIN_PWM_LF1, CH_PWM_L1);
      |   ^~~~~~~~~~~~~
      |   ledcAttach

exit status 1

Compilation error: 'ledcSetup' was not declared in this scope; did you mean 'ledc_stop'?
```

hemangjoshi37a, 2024-10-14:
here is my final working code :
```c
#include <Arduino.h>
#include "driver/ledc.h"

const String robot_type = "DIY_ESP32";
// Required App Version: 0.7.0
// ---------------------------------------------------------------------------
// This Arduino Nano sketch accompanies the OpenBot Android application.
//
// The sketch has the following functionalities:
//  - receive control commands and sensor config from Android application (USB serial)
//  - produce low-level controls (PWM) for the vehicle
//  - toggle left and right indicator signals
//  - wheel odometry based on optical speed sensors
//  - estimate battery voltage via voltage divider
//  - estimate distance based on sonar sensor
//  - control LEDs for status and at front and back of vehicle
//  - send sensor readings to Android application (USB serial)
//  - display vehicle status on OLED
//
// Dependencies: Install via "Tools --> Manage Libraries" (type library name in the search field)
//  - Interrupts: PinChangeInterrupt by Nico Hood (read speed sensors and sonar)
//  - OLED: Adafruit_SSD1306 & Adafruit_GFX (display vehicle status)
//  - Servo: Built-In library by Michael Margolis (required for RC truck)
// Contributors:
//  - October 2020: OLED display support by Ingmar Stapel
//  - December 2021: RC truck support by Usman Fiaz
//  - March 2022: OpenBot-Lite support by William Tan
//  - May 2022: MTV support by Quentin Leboutet
//  - Jan 2023: BLE support by iTinker
// ---------------------------------------------------------------------------

// By Matthias Mueller, 2023
// ---------------------------------------------------------------------------

//------------------------------------------------------//
// DEFINITIONS - DO NOT CHANGE!
//------------------------------------------------------//

//MCUs
#define NANO 328 //Atmega328p
#define ESP32 32 //ESP32

//Robot bodies with Atmega328p as MCU --> Select Arduino Nano as board
#define DIY 0        // DIY without PCB
#define PCB_V1 1     // DIY with PCB V1
#define PCB_V2 2     // DIY with PCB V2
#define RTR_TT 3     // Ready-to-Run with TT-motors
#define RC_CAR 4     // RC truck prototypes
#define LITE 5       // Smaller DIY version for education
//Robot bodies with ESP32 as MCU --> Select ESP32 Dev Module as board
#define RTR_TT2 6   // Ready-to-Run with TT-motors
#define RTR_520 7    // Ready-to-Run with 520-motors
#define MTV 8        // Multi Terrain Vehicle
#define DIY_ESP32 9  // DIY without PCB

//------------------------------------------------------//
// SETUP - Choose your body
//------------------------------------------------------//

// Setup the OpenBot version (DIY, PCB_V1, PCB_V2, RTR_TT, RC_CAR, LITE, RTR_TT2, RTR_520, DIY_ESP32)
#define OPENBOT DIY_ESP32

//------------------------------------------------------//
// SETTINGS - Global settings
//------------------------------------------------------//

// Enable/Disable no phone mode (1,0)
// In no phone mode:
[... 1859 lines cut ...]
    counter_rb++;
  }
}
#endif

#if (HAS_SPEED_SENSORS_MIDDLE)
// ISR: Increment speed sensor counter (left mid)
void update_speed_lm() {
  if (ctrl_left < 0) {
    counter_lm--;
  } else if (ctrl_left > 0) {
    counter_lm++;
  }
}

// ISR: Increment speed sensor counter (right mid)
void update_speed_rm() {
  if (ctrl_right < 0) {
    counter_rm--;
  } else if (ctrl_right > 0) {
    counter_rm++;
  }
}
#endif
```

marshalab, 2024-11-22:
why...does it hurt my eyes so much?

ledcSetup
ledcAttachPin
ledcWrite

At the moment they work differently. It is necessary to rewrite the code related to the PWM output.

thias15, 2024-11-24:
@marshalab, please feel free to submit a PR to improve this!

hemangjoshi37a, 2024-11-25:
@thias15  sure i will once i have some spare time . Should i make a new file for esp32 separately or else how can we integrate this with current code ? because I have done some heavy changes .

hemangjoshi37a, 2024-11-25:
@thias15 apart from this i like all the research you are doing with autonomous drones and all. I wish there is some open research around similar to skydio. I hope you know what skydio is.

nanotuxi, 2024-12-14:
On my forked repo of openbot you will find a branch nanotuxi/arduino-esp32. From there you can download a firmware that compiles on the new arduino-esp32v3+  versions. But be aware that it is untested on real hardware... it just compiles. 
Maybe it can be a good starting point to get things running :-)
You can download the firmware with 
wget  https://raw.githubusercontent.com/nanotuxi/OpenBot/refs/heads/nanotuxi/arduino-esp32/firmware/openbot/openbot.ino
Please report if it helped. Thanks.

marshalab, 2025-01-21:
> On my forked repo of openbot you will find a branch nanotuxi/arduino-esp32. From there you can download a firmware that compiles on the new arduino-esp32v3+ versions. But be aware that it is [...]

Good news, as of 21.01.2025.
Arduino IDE version 2.3.4, with esp 3.1.0 compiles without errors. Moreover, both versions: the original OpenBot v071 and the modified openbot_nanotuxi (from the previous message). I finished assembling the robot, now I need to assemble the firmware and check it.

nanotuxi, 2025-01-21:
> > On my forked repo of openbot you will find a branch nanotuxi/arduino-esp32...
> Good news, as of 21.01.2025. Arduino IDE version 2.3.4, with esp 3.1.0 compiles without errors. Moreover, both versions: the original OpenBot v071 and the modified openbot_nanotuxi (from the previous message). I finished assembling the robot, now I need to assemble the firmware and check it.
> 
> [image]

Be careful with my version. It is untested and there is no garantee that it will work as expected. My advice is to use the original openbot software, as I won't give any support for my version. It was only ment for getting an idea of how things could work😇😉
Have fun.

marshalab, 2025-01-23:
Unfortunately, I was mistaken when I wrote that everything works and compiles. My delight with this project and the confusion in the files (((
But after digging around with the firmware today, I finally launched my robot ESP32_NXT. The main peripherals work, the robot moves. Motors, wheel encoders, ultrasonic sensor, OLED display, all LEDs, Bluetooth connection with a smartphone - work. I only have a front bumper - I roughly rewrote the firmware code. In general, the original firmware turned out to be quite complex. I had to redo a lot of things. Maybe later I'll make a fork and design it beautifully. I need to think about it.

Here is a link to the working firmware for my robot:
[https://gist.githubusercontent.com/marshalab/d90b00b7d50f34f0161a913d0cac239b/raw/8389706d2edd299d5c0f8709c36c2f71c907a6a5/openbot_071mod_marshallab.ino](url)

closed 2025-05-29 by thias15